wheaney/XRLinuxDriver #22: Driver loading without issue but not moving mouse cursor

canguy247, 2023-09-05:
System: GPD Win Max 2 2022 edition
OS: Manjaro Linux
Xreal Air glasses

Xreal_Driver installs and loads without issue and is enabled

Plugged in glasses and no response from mouse cursor on head movement

.xreal_driver_log:
Waiting for glasses
Waiting for glasses
Waiting for glasses

System log just shows:
Started XREAL Air user-space driver.

wheaney, 2023-09-05:
Just want to confirm a few things:
1. `systemctl is-active xreal-air-driver` should print "active."
2. `ps aux | grep xreal | grep root` should return one entry.
3. The glasses are still working as a display

wheaney, 2023-09-05:
I'd also like to see your usb device info for the Airs. Do `sudo lsusb -v -d 3318:0424 > xreal_usb.out` and then attach xreal_usb.out here.

wheaney, 2023-09-05:
Have you confirmed that head tracking works in any other applications (e.g. Nebula, BEAM, etc...)?

canguy247, 2023-09-05:
> Just want to confirm a few things:
> 
>     1. `systemctl is-active xreal-air-driver` should print "active."
> 
>     2. `ps aux | grep xreal | grep root` should return one entry.
> 
>     3. The glasses are still working as a display

1. yes it does
2. yes, just one
3. yes they do

canguy247, 2023-09-05:
> Have you confirmed that head tracking works in any other applications (e.g. Nebula, BEAM, etc...)?

Yes, they work perfectly with Nebula on Android

canguy247, 2023-09-05:
> 

This seems to be the issue.  This device is not detected as usb.  when i do a sudo lsusb it is not on the list at all.

The weird thing is the last time I see anything in the log for this device ID was Sept 1st, which may be the first time I ever plugged it into this laptop...

Sep 01 19:08:38 WinMax2 kernel: usb 7-1: New USB device found, idVendor=3318, idProduct=0424, bcdDevice= 0.01
Sep 01 19:08:38 WinMax2 kernel: hid-generic 0003:3318:0424.0001: hiddev96,hidraw0: USB HID v1.10 Device [Vendor Air] on usb-0000:75:00.3-1/input3
Sep 01 19:08:38 WinMax2 kernel: hid-generic 0003:3318:0424.0002: hiddev97,hidraw1: USB HID v1.10 Device [Vendor Air] on usb-0000:75:00.3-1/input4
Sep 01 19:08:38 WinMax2 kernel: hid-generic 0003:3318:0424.0003: hiddev98,hidraw2: USB HID v1.10 Device [Vendor Air] on usb-0000:75:00.3-1/input5
Sep 01 19:08:41 WinMax2 alsactl[607]: Found hardware: "USB-Audio" "USB Mixer" "USB3318:0424" "" ""

canguy247, 2023-09-05:
This seems to be an issue with the underlying usb system, not with the driver or glasses

closed 2023-09-05 by canguy247