wheaney/XRLinuxDriver #23: [Discussion] Bigger virtual desktop in linux/wayland

kelvie, 2023-09-07:
Didn't see a discussion tab (or a mailing list), so sorry for creating a new issue.

One of the things I want to implement is a larger virtual desktop that follows head tracking. E.g. a 4k virtual desktop that you can pan while moving your head.

Has anyone thought about how one might do this? I can't even figure out how to make a virtual desktop in wayland with edge panning (or is this specific to a compositor like Kwin?)

kelvie, 2023-09-07:
(I'm a developer so I can help with this)

wheaney, 2023-09-07:
I'm actually exploring this right now in [another project](https://github.com/wheaney/breezy-desktop) (FYI, I haven't committed any code right now as I've literally just started). I was inspired by [this proof-of-concept](https://github.com/alexwilson1/nreal_linux_test) project and will probably take a pretty similar approach, though optimized and in a compiled language, with full head-tracking (not just yaw), multiple virtual monitors, the works... ideally anyway. :)

kelvie, 2023-09-07:
From that other project, it still isn't clear how one would do this in Wayland, did you do any research on that part? Or were you also planning to stick with X11?

kelvie, 2023-09-07:
Specifically, how does one even create a virtual desktop in wayland like in X11 using a dummy output.

wheaney, 2023-09-08:
Honestly I wasn't even aware of Wayland as I'm not that deep on linux display managers (or whatever you would call this class of libraries). But I did some searching for "OBS Studio wayland" because I knew the big desktop streaming clients would need to support Wayland capture, and I came across this article: https://itsfoss.com/screen-record-obs-wayland/

Which led me to this code for the OBS pipewire integration: https://github.com/obsproject/obs-studio/tree/3ac08303b546c369f63c1985eff29b0318f1782b/plugins/linux-pipewire

It wouldn't be my first priority to duplicate this as I'd like to just get a POC off the ground with some useful functionality. But maybe adapting that would be your area once I get some code pushed, since I'm not running Wayland.

kelvie, 2023-09-08:
Yeah I think the capture part is easy, not so sure about the virtual
desktop creation part (like, will it need a dummy plug of some sort? Or can
we just create a surface larger than the current desktop)

wheaney, 2023-09-12:
Agreed, I've been digging a lot into wayland because the Steam Deck runs KDE Plasma which I understand the be a Wayland environment. As you said, capturing and rendering the virtual desktop should be fairly easy. I've been searching every resource possible to see what options there are for creating wayland virtual desktop, and this isn't my area of expertise so I don't fully understand what little I have found. For example, [SwayWM](https://github.com/swaywm/sway) can create virtual displays, but that's a full Wayland compositor in its own right, installing that seems pretty heavy-handed and I'm not sure whether two compositors can live alongside each other.

wheaney, 2023-09-12:
[This looks like a solution](https://wayland.app/protocols/kde-output-management-v2) that's unfortunately specific to just KDE Plasma. It looks like they've extended the wayland protocol to allow clients to modify output configurations; something wayland doesn't seem to allow out-of-the-box.

wheaney, 2023-09-12:
It may be that we do this through multiple implementations, depending on the Window System:
* X11 would just use xrandr
* Wayland would depend on the specific compositor:

1. KDE uses the custom protocol specified above
2. GNOME provides a tool called `gsettings` that seems to allow for easily creating a new virtual monitor (e.g. `gsettings set org.gnome.desktop.remote-desktop.rdp screen-share-mode extend` [[source](https://www.reddit.com/r/gnome/comments/uz5as7/gnome_has_made_it_super_simple_to_extend_your/)])

kelvie, 2023-09-12:
Looking at the API it's still not clear how to get a virtual output in plasma, though I just encountered this:

https://www.reddit.com/r/kde/comments/wx8suh/wayland_only_how_to_use_the_new_virtual_output/

I can try to dig around how that works when I have time.

wheaney, 2023-09-12:
I already looked into that on my Steam Deck and couldn't access it, it doesn't have that Screen Sharing app available.

kelvie, 2023-09-12:
I think it's this: https://invent.kde.org/plasma/xdg-desktop-portal-kde/-/blob/master/src/outputsmodel.cpp?ref_type=heads

With the screen chooser here:

https://invent.kde.org/plasma/xdg-desktop-portal-kde/-/blob/master/src/ScreenChooserDialog.qml?ref_type=heads

wheaney, 2023-09-12:
I'm going to close this Issue since it's in the wrong package, and make it [a Discussion in the breezy-desktop repo](https://github.com/wheaney/breezy-desktop/discussions/1).

closed 2023-09-12 by wheaney