wheaney/XRLinuxDriver #34: Alternative joy stick mode

hodasemi, 2023-12-27:
Hi,

it would be nice to an alternative joy stick mode. The current one tracks the velocity and converts it to the right joy stick. My use case is Automobilista 2 where the right joy sticks absolute position is the head rotation inside the car. It would be awesome to look around inside the car, like a light version of VR in this game. Maybe (and hopefully) it is not the only use case for this feature.

wheaney, 2023-12-27:
You're not the first person to request this and it should be pretty easy to implement. I'll work to get it into an upcoming release.

hodasemi, 2023-12-28:
Next week I'll probably have some spare time. I'll try to create a pull request, if that helps. As far as I understand I see 3 things to implement:

* Add a config option for joystick mode
* Add a case check in the output function
* Calibration for the center point. I would probably reuse the part from the stationary mode.

Adding it to the decky ui plugin would be another thing, but on that front I have like zero experience.

hodasemi, 2024-01-14:
I did have way less time than anticipated, but there were a few things that didn't go well:

* The module xrealInterfaceLibrary seems to be private or at least not accessible for me
* I tried to work around it by changing the build files and commenting code. I got it to compile, but then it segfaulted right at the start. 

I may look into it, when time is available.

Sielu-Rooster, 2024-06-28:
Ran into a similar issue in ARMA3, can get the joystick mapped but the velocity vs. absolute position issue makes headtracking with it useless.  Having an 'absolute' mode toggle would be awesome for simming!