wheaney/XRLinuxDriver #42: Unable to update submodules

sjovanovic, 2024-01-19:
I am unable to update submodules because I don't have access to GitLab so I get permission denied.
How to update submodules without having to create a GitLab account?

xM8WVqaG, 2024-01-19:
Replace `git@gitlab.com:` with `https://gitlab.com/` in [`.gitmodules`](https://github.com/wheaney/XRLinuxDriver/blob/main/.gitmodules#L3).
```diff
-	url = git@gitlab.com:TheJackiMonster/nrealAirLinuxDriver.git
+	url = https://gitlab.com/TheJackiMonster/nrealAirLinuxDriver.git
```
The GitLab repo is public, so you don't need to authenticate via SSH to pull it.

sjovanovic, 2024-01-22:
Thanks. After this `git submodule update` command works.

wheaney, 2024-02-05:
This is fixed now.

closed 2024-02-05 by wheaney