wheaney/XRLinuxDriver #48: Inverted Roll

Audiobox, 2024-02-02:
Some situations arise where the roll movement is inverted. Tilting your head left makes the screen tilt right and visa versa. Unfortunately there aren't any settings within game/sim environments that invert this function in a similar manner to y-axis/x-axis inversion. Toggles for axis inversion within the plugin might(?) work

wheaney, 2024-02-02:
Pasting my comments from Discord here:
>i'm not sending roll data
> unless you found a config i have buried in the driver that the decky UI and config script don't expose, you would've had to modify your ocnfig file directly
> so what you're probably seeing is just that the screen rolls with you (or rather, doesn't roll at all, so the screen just "follows" your roll)
> and there's no real way for me to fix that using virtual mouse/joystick
> only way it would be possible would be for me to use a hybrid virtual display + mouse/joystick approach, where the shader is only used for roll, and the mouse is used for x/y movement

wheaney, 2024-02-02:
Going to close this out as it's not really feasible.

closed 2024-02-02 by wheaney