wheaney/XRLinuxDriver #52: Support for Xreal Light glasses

Pandapip1, 2024-02-13:
I imagine this might take a lot of work, so I understand if this ends up as a `wontfix`. I have a pair I can test with.

wheaney, 2024-02-13:
It's definitely something I'd like to tackle, I've even been posting an item for this on my roadmaps. There's a basic integration for them in [this rust package](https://github.com/badicsalex/ar-drivers-rs) but to my recollection it only provides the bare minimum HID integration, not the IMU fusion stuff that's needed to build apps on top of. Not to mention SLAM algorithms needed for the 6DoF. But anyway, I'm okay leaving this open to track it.

Pandapip1, 2024-02-19:
If it helps, I'm working on a Monado driver for them: https://gitlab.freedesktop.org/monado/monado/-/merge_requests/2145