wheaney/XRLinuxDriver #54: Request Support for 120hz mode using Xreal Air

GRYaDad, 2024-03-02:
Xreal Air glasses connect to a device in 60hz mode by default. If user holds the Volume Up rocker until two beeps are heard, this puts the glasses in 120hz mode. When doing this while connected to the Steam Deck, the plug-in no longer recognises that the Xreal Air glasses are plugged in, requiring user to unplug the glasses and plug them back in. Doing so resets the power to the glasses and puts it back in 60hz mode, thus making 120hz mode while using the plug-in currently impossible. It's requested that the plug-in recognizes the switch to 120hz mode in the Xreal Air glasses.

wheaney, 2024-03-14:
Per [this Discord comment](https://discord.com/channels/1156440932846141493/1166820375255056444/1216182349306265710), holding the up rocker may not be switching it the way you're expecting:

> holding the brightness up doesnt switch to 120hz mode
> it switches hdmi audio to usb aduio 
> to select 120 hz u need to manually in settings

Can you confirm?

GRYaDad, 2024-03-17:
Confirmed: your plugin, when changing the refresh rate manually in Steam Deck settings, does support 120hz mode!

Yahoo Mail: Search, Organize, Conquer