wheaney/XRLinuxDriver #61: Udev rules

hodasemi, 2024-06-07:
More like a summary issue resulting from the conversion in the [MR](https://github.com/wheaney/XRLinuxDriver/pull/60)

I'd like to collect some information about the usb devices to build the udev rules.

```bash
# usb-devices

T:  Bus=03 Lev=01 Prnt=07 Port=04 Cnt=01 Dev#= 12 Spd=12   MxCh= 0
D:  Ver= 2.01 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=35ca ProdID=1011 Rev=02.00
S:  Manufacturer=VITURE One
S:  Product=XR Glasses
S:  SerialNumber=2074306C4156
C:  #Ifs= 4 Cfg#= 1 Atr=a0 MxPwr=300mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
E:  Ad=01(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#= 1 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
E:  Ad=02(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#= 2 Alt= 0 #EPs= 1 Cls=02(commc) Sub=02 Prot=01 Driver=cdc_acm
E:  Ad=86(I) Atr=03(Int.) MxPS=   8 Ivl=16ms
I:  If#= 3 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=03(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=85(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
```

```bash
$ ls -la /dev/serial/by-id

lrwxrwxrwx 1 root root 13  7. Jun 12:53 usb-VITURE_One_XR_Glasses_2074306C4156-if02 -> ../../ttyACM0
```

I'm still figuring out the rules

hodasemi, 2024-06-07:
Is there a way to verify the drivers access through the udev rules? I mean some kind of feedback that the driver can access the device correctly

wheaney, 2024-06-07:
Note in the example nreal air rules file, I linked, the `GROUP` should be a group that the run-as user for the systemd service is in, I believe.

wheaney, 2024-06-07:
I believe if we leave off `GROUP` and add `TAG+="uaccess"` to the rule, then it will grant access to all users.

hodasemi, 2024-06-08:
I implemented rules for the viture devices based on what I could find on my system, as stated in the [wiki](https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices):

https://github.com/hodasemi/XRLinuxDriver/commit/69dc1571dcc1c2958a3cbb33642b64ea2afd0893

wheaney, 2024-06-08:
Great job finding that wiki! I did notice this bit, which the current file naming of `80-*` doesn't follow:
> Note: For any rule adding the uaccess tag to be effective, the name of the file it is defined in [has to lexically precede](https://github.com/systemd/systemd/issues/4288#issuecomment-348166161) /usr/lib/udev/rules.d/73-seat-late.rules

hodasemi, 2024-06-08:
Yea I fixed that and verified that I can access the files in /dev/

hodasemi, 2024-06-08:
Can I modify what I see with this driver alone? I mean without the usage of breeze-desktop.

wheaney, 2024-06-08:
> Can I modify what I see

I'm not understanding what you mean with this bit

hodasemi, 2024-06-08:
About the verification process, but as you stated in the MR, I need breezy-desktop for it to change to output.

wheaney, 2024-07-31:
Closing this out since it was merged.

closed 2024-07-31 by wheaney