wheaney/XRLinuxDriver #77: wayland-client not found

mario-aleo, 2024-10-25:
I`ve been trying to install the **xr-driver-git** using yay, but it's yielding an error?

```
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:619 (message):
  The following required packages were not found:

   - wayland-client

Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:841 (_pkg_check_modules_internal)
  CMakeLists.txt:41 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().

```

I have **xorg-xwayland** installed and can't find a package explicit named **wayland-client**.
What am I missing here?

wheaney, 2024-10-25:
Can you try this? `sudo pacman -S wayland`
After that, you may also need `sudo pacman -S wayland-protocols`

Let me know if that resolves it.

mario-aleo, 2024-10-26:
It did worked, that was totally my fault.
But now I'm getting an: missing libudev, I'm trying to install it but without success though I have just started trying.

wheaney, 2024-10-26:
Try `sudo pacman -S systemd-libs and libusb0`

wheaney, 2024-10-26:
So I was just looking at my PKGBUILD and my `depends` list appears to be complete: it will pull in `hidapi` and `libusb` which pull in the `systemd-libs`. Are you doing `makepkg` directly? Or using a tool like `yay`?

wheaney, 2024-10-26:
Just pushed an update to my PKGBUILD to add the wayland dependency. I'd recommend using a tool like `yay` so it does all the dependency management for you.

mario-aleo, 2024-10-26:
To be honest, I'm currently only using yay, I'm not sure why it is downloading some dependencies and missing others :(

wheaney, 2024-10-26:
Can you see if installing `systemd-libs` yourself solves it?

mario-aleo, 2024-10-27:
Sorry for the late reply.
It's working fine now, thanks :)

closed 2024-10-27 by mario-aleo

wheaney, 2024-10-28:
Great to hear it's working now. Let me know which dependencies fixed it.

mario-aleo, 2024-10-28:
systemd-libs

wheaney, 2024-10-28:
I've added that to the dependencies, thanks!