wheaney/XRLinuxDriver #78: Question - USB-c to hdmi video adapter for raspberry pi

TopherTimeMachine, 2024-10-30:
I'd love to run this driver on a raspberry pi 5.  But since the pi doesn't support usb-c video I'd need to run a hdmi usb-c adapter.  

I'm wondering if using a usb-c video splitter cable, if I can still get this driver to work and get the gyro data out of viture pro glasses.

I know I can get the video from the pi to the glasses but I want to make sure I can get the gyro to mouse move movements.

Here is a example cable I was looking at using:
https://www.amazon.com/fairikabe-Converter-Thunberbolt-Compatible-Portable/dp/B0B5XBYQSM

Anyone tried this before or have a recommendation?

wheaney, 2024-10-30:
The amazon page I'm taken to when I click that is the same adapter I use, though I'm not sure why the URL mentions thunderbolt since that's not relevant. So, yes, if that's the one with this description, then it definitely works:
> HDMI to USB C Adapter Cable 4K60Hz 6.6FT, HDMI Source Input to USB Type C Output Display Converter, HDMI 2.0 Compatible with Xreal Air, Nreal Air, Steam Deck Dock, PS5, Xbox, Portable Monitor

Just note that this is a very long and heavy cable, so I found as soon as I got it that I couldn't connect it directly into the glasses because it pulls down pretty hard, causing discomfort and the glasses wanting to tilt up. So I had to buy [another adapter](https://www.amazon.com/gp/product/B08BFS89RG) that would allow me to plug my original glasses cable into the adapter you linked. With that setup, you can get both video and USB data, though I haven't tried this personally on the Pi 5, so just full disclosure I'm not 100% certain it'll work well.

TopherTimeMachine, 2024-11-03:
@wheaney Thanks for the reply.  This sounds promising.  Do you have the problem that some people were saying where the cable adapter gets very hot??

wheaney, 2024-11-05:
I don't use it for long periods very often but I do seem to remember that the adapter gets pretty warm near the HDMI port. Nothing alarming, though.

wheaney, 2025-05-05:
Gonna close this out. Feel free to reopen if anything comes up.

closed 2025-05-05 by wheaney

wheaney, 2026-04-21:
Yes, I believe I used it when testing 6DoF. Note that it has never worked for SBS mode, so it's a no-go if that's important.