wheaney/XRLinuxDriver #82: Invert Y axis option VR Lite mode

pokeparadox, 2025-02-14:
This would be an option to independently invert the mouse y axis when using the VR Lite mode for mouse look.

Some older games either don't allow inversion or tie it into a single setting which doesn't always make sense. This would increase compatibility in these cases.

openmonk, 2025-03-15:
As a workaround you can toggle Y axis of XR mouse using xinput:
```
xr_dev_id=$(xinput list --short | grep "XR virtual mouse" | sed 's/.*id=\([0-9]*\).*/\1/g')
xr_prop_id=$(xinput --list-props $xr_dev_id | grep "Coordinate Transformation Matrix" | sed 's/.*(\(.*\)).*/\1/g')
xinput --set-prop $xr_dev_id $xr_prop_id 1 0 0 0 -1 0 0 0 1
```
Revert back:
```
xinput --set-prop $xr_dev_id $xr_prop_id 1 0 0 0 1 0 0 0 1
```

wheaney, 2025-05-05:
This feature is now available in the latest version (v1.1.0) of the XR Gaming plugin.

closed 2025-05-05 by wheaney