wheaney/XRLinuxDriver #87: RayNeo Air 3s

colinsham, 2025-04-08:
I know I'm probably at the cutting edge here, but I picked up a pair of Air 3s and regret it because there's no XR support.

I've seen in other threads that you ask for lsusb -- attached. It's still 1bbb:af50

Hopefully the SDK you got supports this, and it's just a matter of updating the allowlist.

[air3.txt](https://github.com/user-attachments/files/19641731/air3.txt)

wheaney, 2025-04-08:
I actually already have a local build that works for the 3s. So it should make it to the XR Gaming plugin and Breezy within the next few weeks, I hope.

wheaney, 2025-05-05:
Air 3 support is now in the latest version of the XR Gaming decky plugin. It'll be built into the next Breezy GNOME version probably this week as well.

closed 2025-05-05 by wheaney