wheaney/XRLinuxDriver #88: CMake error when installing xr-driver-git from AUR on Arch Linux

CryZo, 2025-04-10:
Hey there! 👋
I tried installing `xr-driver-git` from the AUR, but I ran into this error:

```
CMake Error at modules/xrealInterfaceLibrary/interface_lib/modules/hidapi/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
```

Any idea how to fix this? I don’t have much experience with CMake, so I’m not sure if just updating the minimum version to 3.5 is the right approach.

wheaney, 2025-04-10:
Thanks for letting me know. It looks like the hidapi repo resolved this issue last year: https://github.com/libusb/hidapi/pull/663

I believe the latest xreal driver already has this pulled in, so I'll just need to update my own submodule reference in order to resolve it.

Chaser-0, 2025-04-10:
Awesome news! Thanks for taking care of it. I can't wait to give breezy a run with my new glasses.

wheaney, 2025-04-10:
v2.0.6 of the driver is in AUR now with the updated hidapi submodule. Can you give it a try?

Chaser-0, 2025-04-10:
That worked like a charm! Thanks for your fast help! 🫶

closed 2025-04-10 by Chaser-0

wheaney, 2025-04-10:
Glad it was that easy. Enjoy!