wheaney/XRLinuxDriver #90: tempfiles causing issue on multiuser system

MartinFlores751, 2025-04-21:
I use multiple users on my arch linux install and ran into an issue caused by this.

You can replicate this by starting breezy-desktop using user A, logging out, and trying to launch the program using user B.

Following is what I found in driver.log:
```
...
2025-04-21 12:57:27.084 Project version: 2.0.6
2025-04-21 12:57:27.084 Driver has been re-enabled
2025-04-21 12:57:27.084 Output mode has been changed to 'external_only'
2025-04-21 12:57:27.090 Using hardware id 4da87fdf7c9b68eab9f19802bb676d72d82bde8a4d9897d1fc7f4ad1b69e4e8d
2025-04-21 12:57:27.094 Feature smooth_follow granted.
2025-04-21 12:57:27.094 Feature sbs granted.
2025-04-21 12:57:27.094 Feature productivity_basic granted.
2025-04-21 12:57:27.094 Starting up XR driver
2025-04-21 12:57:27.095 Waiting for glasses
2025-04-21 12:57:27.095 Breezy desktop has been enabled
2025-04-21 12:57:27.098 Found device with vendor ID 0x35ca and product ID 0x101d
2025-04-21 12:57:27.098 Found device with vendor ID 0x35ca and product ID 0x101d
2025-04-21 12:57:29.132 Device connected, redirecting input to external_only...
2025-04-21 12:57:29.132 [ERROR] Could not create IPC shared file
2025-04-21 12:57:29.332 Project version: 2.0.6
2025-04-21 12:57:29.332 Driver has been re-enabled
2025-04-21 12:57:29.332 Output mode has been changed to 'external_only'
2025-04-21 12:57:29.338 Using hardware id 4da87fdf7c9b68eab9f19802bb676d72d82bde8a4d9897d1fc7f4ad1b69e4e8d
2025-04-21 12:57:29.341 Feature smooth_follow granted.
2025-04-21 12:57:29.342 Feature sbs granted.
2025-04-21 12:57:29.342 Feature productivity_basic granted.
2025-04-21 12:57:29.342 Starting up XR driver
2025-04-21 12:57:29.342 Waiting for glasses
2025-04-21 12:57:29.343 Breezy desktop has been enabled
2025-04-21 12:57:29.345 Found device with vendor ID 0x35ca and product ID 0x101d
2025-04-21 12:57:29.345 Found device with vendor ID 0x35ca and product ID 0x101d
2025-04-21 12:57:31.376 Device connected, redirecting input to external_only...
2025-04-21 12:57:31.376 [ERROR] Could not create IPC shared file
2025-04-21 12:57:31.582 Project version: 2.0.6
2025-04-21 12:57:31.582 Driver has been re-enabled
2025-04-21 12:57:31.582 Output mode has been changed to 'external_only'
2025-04-21 12:57:31.587 Using hardware id 4da87fdf7c9b68eab9f19802bb676d72d82bde8a4d9897d1fc7f4ad1b69e4e8d
2025-04-21 12:57:31.591 Feature smooth_follow granted.
2025-04-21 12:57:31.591 Feature sbs granted.
2025-04-21 12:57:31.591 Feature productivity_basic granted.
2025-04-21 12:57:31.591 Starting up XR driver
2025-04-21 12:57:31.592 Waiting for glasses
2025-04-21 12:57:31.593 Breezy desktop has been enabled
2025-04-21 12:57:31.595 Found device with vendor ID 0x35ca and product ID 0x101d
2025-04-21 12:57:31.595 Found device with vendor ID 0x35ca and product ID 0x101d
2025-04-21 12:57:33.597 Waiting for glasses
```

I found the `[ERROR]` string in `ipc.c`:
```C
void setup_ipc_value(const char *name, void **shmemValue, size_t size, bool debug) {
    char *path = malloc(strlen(sombrero_ipc_file_prefix) + strlen(name) + 1);
    strcpy(path, sombrero_ipc_file_prefix);
    strcat(path, name);

    FILE *ipc_file = fopen(path, "w");
    if (ipc_file == NULL) {
        log_error("Could not create IPC shared file\n");
        exit(1);
    }
```

And then found the file prefix in the same file:
```C
const char *sombrero_ipc_file_prefix = "/tmp/shader_runtime_";
```

Doing a quick check using `ls -l /tmp/shader_runtime_*` showed the files existing and being owned by user A. Deleting the files will allow user B to run the program.

I'm not sure if it cleaning up the files on logout can be done, or if scoping based on users is practical, but any solution to this would be appreciated!

wheaney, 2025-04-27:
Thanks for bringing this to my attention, I'm surprised it hasn't come up before. Deleting it would work but doesn't seem like it would be reliable enough (in the case of a crash or some other case where the deletion doesn't occur), but what I think would work is giving the files global read/write.

wheaney, 2025-05-14:
This should be resolved now in Breezy Vulkan (XR Gaming) and Breezy GNOME.

closed 2025-05-14 by wheaney