wheaney/XRLinuxDriver #92: question

MR-Geri, 2025-05-14:
The current version has support for xreal one? If there is support, at what level? How many virtual screens can I create? Does tracking the screen position in space work?

wheaney, 2025-05-14:
XREAL is very anti-developer and doesn't provide general-purpose SDKs for their hardware, nor do they respond to inquiries asking for help in building/maintaining them. The XREAL Air series of glasses was reverse engineered and an open source driver was written by the community. The Ones have a different protocol and would need to go through a similar process of reverse engineering. That hasn't been done yet, and I don't know if/when it will be done.

For future inquiries about supported hardware, please reference the [Supported Devices](https://github.com/wheaney/XRLinuxDriver#supported-devices) section of the README.

closed 2025-05-14 by wheaney

wheaney, 2025-06-21:
No, my driver doesn't have a gyro output mode.