wheaney/XRLinuxDriver #94: Viture Luma Pro

lukecharman, 2025-07-18:
Hi, I got my Luma Pros, and the plugin isn't compatible yet. I'm a dev too so I'm not gonna complain, you're a human with a life. Is there anything I can do to help with the process of enabling them? Happy to open a PR or at least have a play with things to see if I can do it, I'm more of a mobile developer but happy to learn :)

THFourteen, 2025-07-22:
Tried it briefly just now, will try it a bit more over the next couple of days - initial feedback : 

- Normal Follow mode works as expected - am able to move the screen top/bottom/left/right and adjust the size of the screen no problems
- Smooth follow mode doesn't work as expected, screen jumps around and moves of its own accord
- Curved Display doesn't seem to have a discernible impact
- Virtual Display exhibits similar behaviour to smooth follow mode

wheaney, 2025-07-22:
Yep, exactly. Don't use the re-center functionality and see where the screen ends up after 10 minutes or more.

THFourteen, 2025-07-23:
So i played for an hour or so this morning and i _think_ the screen is moving over time in virtual display mode. its quite hard to remember exactly where i started with my head, but i feel like after 10-20mins it was definitely not where i left it.

Auto centering was switched off

To be honest its a minor issue at this point, the app is totally useable in current state with Luma Pros!

wheaney, 2025-07-25:
I've received my pair of glasses and made some tweaks that I think will address the biggest issues for the Lumas. I've updated the instructions above with the v1.2.2 binary. Note that you should update to the latest firmware and run the calibration tool from the VITURE firmware update website.

wheaney, 2025-07-26:
That's too bad about the drift. Unfortunately it's not something the plugin has any control over. I would recommend setting down the glasses for a minute after plugging in just in case their algorithm is doing some continuous background calibration.

closed 2025-12-03 by wheaney